dfu-programmer/dfu-programmer #15: Query - erase - application space and bootloader space separate on atxmega

arc12, 2015-01-03:
Hi -
not sure if this is an issue, but I thought an atxmega32a4u had 32k application space and a separate 4k bootloader space.
However executing an erase leads to a message:
"Checking memory from 0x0 to 0x6fff", which is NOT the full 32k.

Cheers, Adam

skirridsystems, 2015-01-05:
Hi Adam,

Dammit you're right! I'm amazed nobody has noticed this before.
And sod's law, your email came in about an hour after I cut the 0.7.1
release. Doh!

It seems the XMega is different from the AVR and AVR32 which both include
the bootloader area within the main flash area, so the size of the
bootloader is subtracted from the total flash size. Not so with the XMega.
The faulty code is around line 437 of arguments.c. I'll take a look at
fixing it.

Simon

skirridsystems, 2015-01-05:
Pull request #16 addresses this issue. If you are able to test it I would be grateful.

arc12, 2015-01-05:
Happy to, although my task starts at line 1 of "Build procedure for Windows", so if you happen to have a compiled vsn to hand...

Adam

skirridsystems, 2015-01-05:
Do you have a private mail address I can send it to?
Simon

arc12, 2015-01-10:
Tested against ATxmega32AU4 for erase/write/verify cycle.
Works as expected.
Adam

closed 2022-10-05 by skirridsystems