dfu-programmer/dfu-programmer #25: specifying the bus and device doesn't work

feldim2425, 2015-09-26:
If I specify the bus and device number it doesn't find the chip.

`$  lsusb
Bus 004 Device 002: ID 8087:8001 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:8009 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 2109:0810  
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 03eb:2ff4 Atmel Corp. 
Bus 001 Device 006: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB`

`$ sudo dfu-programmer atmega32u4:1,12 get bootloader-version
dfu-programmer: no device present.`

but if i type:
`$ sudo dfu-programmer atmega32u4 get bootloader-version
Bootloader Version: 0x00 (0)`

swinman, 2015-10-02:
I just tried this and can't replicate your error...  Mine works when i use
the bus/device id

swinman, 2015-10-02:
Can you provide some information, maybe also use --debug=200

feldim2425, 2015-10-02:
I installed dfu-programmer with  apt-get install , but that has downloaded an older version (0.7.0). It works fine in 0.7.2

closed 2015-10-02 by feldim2425

cinderblock, 2022-10-25:
This is probably caused by you build linking to libusb0 which uses a different numbering scheme than libusb1. We've since removed support for libusb0 to prevent and future confusion.