dfu-programmer/dfu-programmer #26: Handling of write locked Application_Table section of ATXMEGA devices

N-Bz, 2016-01-12:
Hi,

I'm working on an atxmega32c4, but from reading the code I'm pretty sure that the issue will exist for other devices with this flash size.

I use the Application_Table as an extension of the Bootloader section (i.e. all dfu-related stuff is in the Bootloader section, but more init code is present in the Application_Table), and this code must be protected against erases/flash. (Typically, this means that our BLBB and BLBAT lock bits are set to "WLOCK", while the BLBA and LB are left to NOLOCK).

In this case, I have the following issues:

1. The erase command does not work (but this seem unrelated to dfu-programmer, and needs to be checked from the bootloader, as it should still erase the Application section)
2. The blank check always return an error, because obviously the Application_Table section is never blank, so each flash action requires --force.
3. After flashing, the validate fails because the Application_Table section is still not blank (thus different to what is requested in my .hex file). The validate fails only on data outside region (so the actual flash works).

I've developed a few workarounds on my fork of the project (basically I added a command line flag telling to ignore "outside region" errors while flashing), but I'm pretty sure that a more clean handling could be done (typically, by ignoring only the errors in the Application_Table section if a given flag is set, instead of ignoring all errors). If you're willing to integrate my changes, I can clean them and make a pull request.

Regards,
Nicolas.

swinman, 2016-01-18:
Nicolas,

Go ahead... I get similar out range validation errors and use the return
code to suppress the error.  I like the idea of having a flag that limits
the validation to within region.

-walker

skirridsystems, 2022-10-10:
Closed after 6 years with no further action

closed 2022-10-10 by skirridsystems

cinderblock, 2022-10-10:
Might be fixed by #28 if/when that lands