dfu-programmer/dfu-programmer #29: Issue with erase and flash for atxmega128a3u

dwsobw, 2016-02-13:
Hi,

I am not sure if this is related to  #26, but I have also problems erasing and flashing an XMega
with dfu-programmer (git head/5254ee and version 0.6.1).
But my bootloader seems to be fine, because Atmel's programmer Flip (I am using Version 3.4.7) from an XP Virtual Machine can successfully erase and flash.

I am not sure if it matters that Flip does more steps (open, erase, blank check, program, verify) than dfu-programmer (erase, flash). 

But Flip submits different data to flash.
Flip ([flip_allsteps_usbdump.txt](https://github.com/dfu-programmer/dfu-programmer/files/128944/flip_allsteps_usbdump.txt), line 190) starts with "698: 0100 0000 0279"
while dfu-programmer ([dfuprogrammer_eraseflash_usbdump.txt](https://github.com/dfu-programmer/dfu-programmer/files/128945/dfuprogrammer_eraseflash_usbdump.txt), line 82) starts with "1104: 0100 0000 03ff".

Also Flip's data stops much earlier (last common: ffcf) while dfu-programmer continues with tons of "ffff" and finishes with "0000 0000 1044 4655 0110 ffff ffff ffff".

I have also attached the debug output of dfu-programmer ([dfuprogrammer_eraseflash_debugoutput.txt](https://github.com/dfu-programmer/dfu-programmer/files/128946/dfuprogrammer_eraseflash_debugoutput.txt))

I am pretty clueless what this all means, but if I can provide some more information / try something that might help resolve this issue please let me know.

Thanks for your awesome software!
Daniel

[flip_allsteps_usbdump.txt](https://github.com/dfu-programmer/dfu-programmer/files/128944/flip_allsteps_usbdump.txt)
[dfuprogrammer_eraseflash_usbdump.txt](https://github.com/dfu-programmer/dfu-programmer/files/128945/dfuprogrammer_eraseflash_usbdump.txt)
[dfuprogrammer_eraseflash_debugoutput.txt](https://github.com/dfu-programmer/dfu-programmer/files/128946/dfuprogrammer_eraseflash_debugoutput.txt)

dwsobw, 2016-02-13:
I am not sure if it is relevant but here is the hex file I am trying to flash:
[blinker.hex.txt](https://github.com/dfu-programmer/dfu-programmer/files/128950/blinker.hex.txt)

dwsobw, 2016-02-13:
The path to #26 does not help.

swinman, 2016-02-13:
What commands are you using and what is the error?

Use the debug flag to get more info about errors.  The debug level below
100 give dfu-programmer info and over 100 starts to add libusb info

dwsobw, 2016-02-13:
./dfu-programmer atxmega128a3u erase --debug 999999
./dfu-programmer atxmega128a3u flash /home/dw/dev/xmega/blinker.hex --debug 999999

USB error: error sending control message: Broken pipe
dfu.c:892: dfu_download ERR: -EPIPE: a) Babble detect or b) Endpoint stalled 0xffffffe0 (-32)
Device is write protected.
dfu.c:260: dfu_clear_status( 0xbfbc5b18 )
atmel.c:1292: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:514: Error writing memory data. (err -4)

[dfuprogrammer_eraseflash_debugoutput.txt](https://github.com/dfu-programmer/dfu-programmer/files/128946/dfuprogrammer_eraseflash_debugoutput.txt)

dwsobw, 2016-02-13:
I have also played a bit with the code. 
If I out commented the footer in __atmel_flash_block and I out commented the determine the limits of where actual data resides in the buffer for-loop in atmel_flash. 
Then I get the same flash data. 
So I would guess that the block-wise flashing does not make a difference, but Flip apparently does not use a footer.
My guess is that it is somehow a difference in what is erased.

[flip_allsteps_usbdump.txt](https://github.com/dfu-programmer/dfu-programmer/files/128944/flip_allsteps_usbdump.txt)
[dfuprogrammer_eraseflash_usbdump.txt](https://github.com/dfu-programmer/dfu-programmer/files/128945/dfuprogrammer_eraseflash_usbdump.txt)

Contain the usb dump, i.e. data send via usb. As far as i can tell the first number is always the number of bytes sent. 
If you look for the sections which start with "erase" in both files you see slight differences, but I havent figured out why.

dfu-progammer:
`erase
  0.000000 -->0 
  0.001223 -->0 
  0.001320 -->0 
  0.002222 -->0 
  0.002328 0<-- 
  0.003221 0<-- 6: 0f00 0000 0a00
  0.003367 -->0 
  0.004222 -->0 
  0.004292 0<-- 
  0.005221 0<-- 6: 0000 0000 0200
  0.005457 -->0 4: 0603 0000
  0.006220 -->0 
  0.006290 0<-- 
  0.007222 0<-- 6: 0000 0000 0200
  0.007436 -->0 5: 0603 0100 00
  0.008220 -->0 
  0.008289 0<-- 
  0.009221 0<-- 6: 0000 0000 0200
  0.009424 -->0 6: 0301 0000 ffff
  0.011192 -->0 
  0.011292 0<-- 
  0.012193 0<-- 6: 0500 0000 0200
  0.012373 0<-- 
  0.013193 0<-- 2: 0000
  0.013404 -->0 3: 0400 ff
  0.020194 -->0 
  0.020285 0<-- 
  0.021194 0<-- 6: 0000 0000 0200
  0.021451 -->0 4: 0603 0000
  0.022191 -->0 
  0.022284 0<-- 
  0.023193 0<-- 6: 0000 0000 0200
  0.023414 -->0 5: 0603 0100 00
  0.024192 -->0 
  0.024287 0<-- 
  0.025192 0<-- 6: 0000 0000 0200
  0.025428 -->0 6: 0301 0000 ffff
  0.244224 -->0 
  0.244322 0<-- 
  0.245222 0<-- 6: 0000 0000 0200
  0.245468 -->0 5: 0603 0100 01
  0.246222 -->0 
  0.246295 0<-- 
  0.247220 0<-- 6: 0000 0000 0200
  0.247429 -->0 6: 0301 0000 ffff
  0.466227 -->0 
  0.466312 0<-- 
  0.467222 0<-- 6: 0000 0000 0200
`

Flip:
`erase
 10.410565 -->0 4: 0603 0000
 10.412423 -->0 
 10.413551 0<-- 
 10.414417 0<-- 6: 0000 0000 0200
 10.415597 -->0 3: 0400 ff
 10.422417 -->0 
 10.424610 0<-- 
 10.425421 0<-- 6: 0000 0000 0200
 10.437610 -->0 4: 0603 0005
 10.439427 -->0 
 10.440192 0<-- 
 10.440405 0<-- 6: 0000 0000 0200
 10.442111 -->0 6: 0300 0000 0000
 10.442417 -->0 
 10.444275 0<-- 
 10.445401 0<-- 6: 0000 0000 0200
 10.447073 0<-- 
 10.447398 0<-- 1: 1e
 10.450113 -->0 4: 0603 0005
 10.450397 -->0 
 10.452124 0<-- 
 10.452401 0<-- 6: 0000 0000 0200
 10.454151 -->0 6: 0300 0001 0001
 10.454401 -->0 
 10.456030 0<-- 
 10.456401 0<-- 6: 0000 0000 0200
 10.458060 0<-- 
 10.458400 0<-- 1: 97
 10.462133 -->0 4: 0603 0005
 10.462401 -->0 
 10.464146 0<-- 
 10.464401 0<-- 6: 0000 0000 0200
 10.466173 -->0 6: 0300 0002 0002
 10.466401 -->0 
 10.468062 0<-- 
 10.468401 0<-- 6: 0000 0000 0200
 10.470195 0<-- 
 10.470400 0<-- 1: 42
 10.473269 -->0 4: 0603 0005
 10.474403 -->0 
 10.476154 0<-- 
 10.476418 0<-- 6: 0000 0000 0200
 10.478192 -->0 6: 0300 0003 0003
 10.478418 -->0 
 10.480039 0<-- 
 10.480402 0<-- 6: 0000 0000 0200
 10.482049 0<-- 
 10.482402 0<-- 1: 08
 10.487378 -->0 4: 0603 0004
 10.488401 -->0 
 10.490141 0<-- 
 10.490408 0<-- 6: 0000 0000 0200
 10.492070 -->0 6: 0300 0001 0001
 10.492405 -->0 
 10.494095 0<-- 
 10.494405 0<-- 6: 0000 0000 0200
 10.496130 0<-- 
 10.496405 0<-- 1: 00
 10.500396 -->0 4: 0603 0004
 10.501408 -->0 
 10.503401 0<-- 
 10.504406 0<-- 6: 0000 0000 0200
 10.506082 -->0 6: 0300 0002 0002
 10.506406 -->0 
 10.508057 0<-- 
 10.508419 0<-- 6: 0000 0000 0200
 10.510078 0<-- 
 10.510402 0<-- 1: 00
 10.513135 -->0 4: 0603 0004
 10.513408 -->0 
 10.515118 0<-- 
 10.515397 0<-- 6: 0000 0000 0200
 10.517053 -->0 6: 0300 0000 0000
 10.517403 -->0 
 10.519078 0<-- 
 10.519405 0<-- 6: 0000 0000 0200
 10.521123 0<-- 
 10.521418 0<-- 1: 04
 10.523279 -->0 4: 0603 0000
 10.524417 -->0 
 10.526162 0<-- 
 10.526402 0<-- 6: 0000 0000 0200
`

SaubaSogI, 2018-03-12:
Hello,

Was there an solution to fix this?
We've the same issue with the ATXMEGA128a3u. 
Errormessage:

> dfu.c:961: Unknown error 0xfffffff7 (-9)

We've tested this on Linux ARM (dfu-programmer versions 0.6.1 - 0.7.2) and Linux x86 environment (Ubuntu, dfu-programmer version 0.6.1).
Application 'Flip' is working fine.

We're calling the application in following way:

> DEBUGLEVEL=999999
> TARGET=atxmega128a3u
> FW_FILE=DL_3.50_640T_V1.5.hex
> #FW_FILE=FW_3v51_ErrorLogger.hex
> 
> sudo dfu-programmer $TARGET erase --debug $DEBUGLEVEL
> echo "------------------------------------------"
> echo "Return from erasing: $?"
> echo "------------------------------------------"
> 
> sudo dfu-programmer $TARGET flash $FW_FILE --debug $DEBUGLEVEL
> echo "------------------------------------------"
> echo "Return from flashing: $?"
> echo "------------------------------------------"
> sudo dfu-programmer $TARGET start

Attached the extended log file
[dfu-programmer.ext.debug.log](https://github.com/dfu-programmer/dfu-programmer/files/1803421/dfu-programmer.ext.debug.log)

Can somebody help us?
Best Regards,
Helmut

cinderblock, 2022-10-21:
Hey @SaubaSogI, I think you may have a different issue entirely. Can you please move your discussion to #68?

---

@dwsobw Are you still having this issue?

I see "`Device is write protected.`" printed in your logs. Is your device full or does it have lock bits set?

cinderblock, 2026-11-12:
@dwsobw A lot has changed since 0.7.2. v1.1.0 added `flash --erase-first`, which erases and flashes in a single session, the way Flip does. Could you try

```
dfu-programmer atxmega128a3u flash --erase-first blinker.hex
```

with v1.1.0 and let us know if it still reports "Device is write protected"? If we don't hear back in a few weeks we'll close this. Feel free to reopen.