dfu-programmer/dfu-programmer #30: Wish: Erase user page

jans23, 2016-07-18:
AFAIK dfu-programmer's "erase" command does not erase the user page of AVR32. It would be great to let dfu-programmer also erase the user page (perhaps when executed with a certain option), similarly to what "batchisp" can do.

swinman, 2016-07-20:
You are correct, erase does not erase the user page.  The last 4 bytes of
the user page (at least for at32uc3b) contain configuration information
that is used by the chip (for example the pin number to check on startup to
determine if the chip should start in boot mode) see
http://www.atmel.com/images/doc7745.pdf

If you erase the user page, you should make sure that this word doesn't get
erased (unless you really want to do that).

One way to erase the user page is with the following commands

dfu-programmer <target> read --user > user.hex

   then edit user.hex manually to clear whatever bits you wish to clear
(make sure you update the hex checksums appropriately, see
https://en.wikipedia.org/wiki/Intel_HEX).. Unlike the regular flash,
writing a new user page doesn't require you to erase the page first.  I
would make a backup copy of the user page before you edit it.  Once you are
ready you can write the new user page

dfu-programmer <target> flash --user --force user.hex

-walker

M4iKK, 2017-09-29:
@swinman already said everthing. Let's close this?

skirridsystems, 2017-09-29:
Agreed, but add the workaround to the documentation.
Simon

closed 2022-10-10 by skirridsystems