dfu-programmer/dfu-programmer #33: No device present

orangeudav, 2016-11-20:
I try to flash ATmega16U2 (in Rumba board) with dfu-programmer running on mac os 10.11.6.

```
$ lsusb
Bus 020 Device 031: ID 03eb:204b Atmel Corporation RUMBA - ATmega 2560 co  Serial: 5533033373035130B171
```
I changed VID in the source code, but still getting error
```
sudo ~/not-my-projects/dfu-programmer/src/dfu-programmer atmega16u2:20,31 flash RUMBA_16U2_USB2Serial.hex --force --debug 99999999
     target: atmega16u2
    chip_id: 0x204b
  vendor_id: 0x03eb
    command: flash
      quiet: false
      debug: 99999999
device_type: AVR
------ command specific below ------
   validate: true
   hex file: RUMBA_16U2_USB2Serial.hex

dfu.c:422: dfu_device_init( 1003, 8267, 0x7fff543788b0, true, false )
dfu-programmer: no device present.
```

Whytehorse, 2017-01-06:
same here on Ubuntu 16.04 
```
$ sudo dfu-programmer atmega16u2 get ID1
dfu-programmer: no device present.
```

swinman, 2017-03-22:
Orangeudav -- specifying bus and device using libusb 0 does not work as
expected because the bus and device that are reported when you use `lsusb`
is different from what the code for libusb0 checks.  It seems that your
version of dfu-programmer is using libusb0.  I've added some additional
debugging info in a branch called libusb_no_device in my personal
dfu-programmer fork (github.com/swinman/dfu-programmer).  It prints info
for each device that it finds and includes the bus/port number calculates.
If you do the same debugging test with that version it should give you more
info.

In general, I'm not a fan of libusb0 and use dfu-programmer with libusb1 on
ubuntu and windows.

-walker

starock, 2017-09-29:
MacOS 10.12 +1

maratmkhitaryan, 2018-02-11:
If you using Arduino, add again jumper and try flash it :). Works for me.

avsingh999, 2018-05-22:
I have the same problem Anyone solved this  problem

bklimt, 2018-06-03:
The original poster has a different problem than @pc-magas and maybe some other commenters. You're probably using the wrong target. For example, with lsusb you see:

`Bus 001 Device 058: ID 03eb:2ff7 Atmel Corp. `

The vendor id is 03eb and the product id is 2ff7. If you look at the targets, you'll see that this id corresponds to `at90usb82`, not `atmega8u2`.

https://github.com/dfu-programmer/dfu-programmer/blob/5254ee489f0048c36f1af660f2354a7746b3f8cc/src/arguments.c#L106

So, at least for @pc-magas, changing the command thus should fix it:

`sudo dfu-programmer at90usb82 erase`

ogallagher, 2019-06-18:
I was able to solve the `dfu-programmer: no device present` issue by following the official Arduino DFU programming guide [here](https://www.arduino.cc/en/Hacking/DFUProgramming8U2).

The key step was to "Reset the 8u2 or 16u2" before trying to erase the (in my case) atmega16u2 chip, by briefly connecting the `RESET` and `GND` pins as shown:

[image]

sf750086, 2019-08-12:
Hello guys,
I have same problem no device found.
i am using win 7, latest dfu version, UNO r3 board (m16u2).
When i try to erase i got the error no device found.
i tried every possible thing on web. yet my problem is solved.

knezi, 2020-05-13:
I get the same problem and it seems like a problem with the Atmega16u2 (firmware or hardware?). It is the same, as described here https://github.com/tessel/node-usb/issues/211#issuecomment-350325695_.
Basically, the usb connection to 16u2 breaks everytime libusb tries to open it. The chip connects immediately again, but gets a new address. Hence it looks like the device is still connected but inaccesible.

You can check this by looking at `dmesg`. Everytime I run `dfu-programmer` the devices disconnects and connects and `lsusb` increases the address by one.

The only thing I can think of is to use Arduino to flash it manually or connect to serial port.

ShadeTechnik, 2020-08-28:
I had the same issue, it's a Linux kernel autosuspend issue. See this post:
[https://sourceforge.net/p/dfu-programmer/mailman/message/504454/](url)
But that is not the solution that worked for me, this is...answer#3:
[https://unix.stackexchange.com/questions/91027/how-to-disable-usb-autosuspend-on-kernel-3-7-10-or-above](url)

Creastoff, 2022-07-31:
I had no end of issues interacting with an atmega16u2 (Arduino Uno R3) on Windows 10.
I spun up a Ubuntu VM and forwarded the DFU device (see bridging RESET & GND earlier in this post) and was able to interact with the chip.

Curious, I detached from my VM and ran the following Powershell script which indicated there was an error on the device.

```
$productId = '2fef'
Get-PnpDevice -PresentOnly | Where-Object { $_.InstanceId -match $productId }
Status     Class           FriendlyName                                                                     InstanceId     
------     -----           ------------                                                                     ----------     
Error                                                                                                       USB\VID_03EB&P…
```

closed 2022-10-24 by cinderblock