dfu-programmer/dfu-programmer #36: Add support for at32u4 chips

pataelmo, 2016-12-07:
It would be nice if this also worked for the newer chips being used in the Arduino micro boards.  Let me know if anyone has tested this or if there is simple work to add support I'll be happy to help.

Thanks!

skirridsystems, 2016-12-08:
ATMega32U4 should already be supported. Is it not working? I don't have
anything to test on.
Simon

pataelmo, 2016-12-09:
Sorry, I haven't tested yet, I was just looking for support for those chips and they were not on the supported list.

I will update this once I can test it out.  If it works then this becomes a documentation bug.

skirridsystems, 2016-12-09:
Where is it not documented? It is listed by the program with --targets and
it appears on the website:
AVR based controllers:

at90usb1287, at90usb1286, at90usb1287-4k, at90usb1286-4k, at90usb647,
at90usb646, at90usb162, at90usb82, atmega32u6, atmega32u4, atmega32u2,
atmega16u4, atmega16u2, atmega8u2
Have I missed somewhere?

Simon

pataelmo, 2016-12-09:
I'm sorry, this is a me not paying close enough attention to the group -> device naming conventions.  I don't know where I saw them as AT32U4's, but until I looked at the chip I didn't realize the actual part is the ATMEGA32U4.

That plus me not looking closely at the supported list, meant I totally missed that.   So, you are correct it is listed as supported.  Thank you for working on this awesome project.  And for being patient with me missing this.

closed 2016-12-09 by pataelmo