dfu-programmer/dfu-programmer #39: Split-off a shared library for use in front-ends

hadess, 2017-06-07:
This would make it easier to use front-ends to update firmwares, especially for devices with wider commercial availability. In my case, I'd like to add a firmware upgrade facility for the Retrode, which uses an AVR chip internally.

See https://github.com/hughsie/fwupd/issues/128

hadess, 2017-06-08:
@hughsie what would be the minimal API to implement to make use of it in https://github.com/hughsie/fwupd/tree/master/libdfu, or would we need a separate library with the same external API footprint?

hughsie, 2017-06-08:
@hadess I think it would be okay to just have *anything* e.g. a simple libdfuavr and then we can just write a simple plugin around that. The best way obviously would be to contribute the stm32-specific bits to the fwupd libdfu. I guess it depends on what the dfu-programmer team want to do.

hadess, 2017-06-08:
I'm mostly wondering about what the external API is supposed to offer (detecting the firmware format, detecting the firmware version, flashing a specific device given a sysfs path with callbacks for progress?).

EspoTek, 2017-07-07:
I'm doing exactly this right now!
It's a little spinoff called "libdfuprog".

Works exactly the same as the standalone exe but you pass your command line arguments as a string to a function rather than over the command line.
You could also access any of the internal functions, in theory.  But I haven't tested this because I don't understand dfu-programmer's internals.

Have tried (and failed) to get the .DLL build working on Windows (apparently this is due to MINGW and MSVC using different DLL types), but it's up and running on Mac right now as a .dylib.  Have erased an AVR chip with it and all.  Moving to Linux (and Android!!) now.  Will push the build scripts and usage instructions when I'm done.

cinderblock, 2022-10-25:
See #72