dfu-programmer/dfu-programmer #42: WORD1 forced to be E11EFFD7

thekenu, 2017-09-25:
The following observations were tested on Windows 10 and Debian 9 for AT32UC3A3256 and AT32UC3C2512:

I noticed that after if I use flash, erase, getfuse, or readfuse, my WORD1 always gets set to **E11EFFD7** afterwards, which renders my microcontroller useless because WORD1 = E11EFFD7 means that ISP_FORCE = 1 and the microcontroller will be forced into DFU mode no matter what. 

I did some debugging using GDB and narrowed down the problem to the dfu_transfer_out() function. That is, before dfu_transfer_out() my WORD1 is a pre-defined value (e.g. E11EFDD9) and after dfu_transfer_out WORD1 is now  **E11EFFD7**.

thekenu, 2017-09-26:
I want to also note that I am using the AVR USB DFU Bootloader Version 1.1.0

skirridsystems, 2017-09-27:
It's possible this is a problem with the bootloader rather than dfu-prog.
Simon

thekenu, 2017-09-27:
The bootloader consistently works fine with ATMEL Flip (BatchISP, to be exact) to flash and erase programs, which leads me to believe the bootloader is not the problem. Thoughts?

skirridsystems, 2017-09-27:
Ah, fair comment.
There is no readfuse command, but getfuse should be benign. Where is
dfu_command_out being called from at the time it corrupts WORD1? What
parameters is it being passed?

Simon

thekenu, 2017-09-27:
**readfuse**  is a BatchISP term and I simply got it mixed up with **getfuse** :-) Calling **dfu_transfer_out** from a few different functions all seem to corrupt WORD1, but I will focus on **getfuse**  as an example. 

Here are the parameters I passed:
```
dfu-programmer at32uc3a3256  getfuse ISP_FORCE
```
The first call to **dfu_transfer_out** already corrupts WORD1, and the parameters being passed are as follow:
```
device =  {handle = 0x961e70, interface = 0, type = 4}
request = 1
value = 0
*data = 0x64fd4c 
length = 4
```
Additionally, the first four bytes by *data (@ 0x64fd4c) are 6,3,0,0.

skirridsystems, 2017-09-27:
That doesn't seem quite right, following through the code manually.
execute_getfuse
  security_check
    atmel_getsecure
      dfu_clear_status
        dfu_transfer_out(device, 4, 0, NULL, 0);
      dfu_download(device, 4, [6, 3, 0, 2]);
        dfu_transfer_out(device, 1, 0, [6, 3, 0, 2], 4);
    atmel_read_fuses
      atmel_select_memory_unit
        dfu_download(device, 4, [6, 3, 0, 3]);
          dfu_transfer_out(device, 1, 0, [6, 3, 0, 3], 4);

It goes on, but none of the first 3 calls to dfu_transfer_out use the data
you describe. Or have I missed a call somewhere?

Simon

thekenu, 2017-09-27:
Your

```
dfu_download(device, 4, [6, 3, 0, 2]);
dfu_transfer_out(device, 1, 0, [6, 3, 0, 2], 4);
```
matches exactly with my

```
#0  dfu_transfer_out (device=0x64ff04, request=1 '\001', value=0, data=0x64fd4c "\006\003", length=4)
    at dfu.c:758
#1  0x004095af in dfu_download (device=0x64ff04, length=4, data=0x64fd4c "\006\003") at dfu.c:178
```
I can also print out data to confirm
```
(gdb) x/4ub 0x64fd4c
0x64fd4c:       6       3       0       2
```

I don't see a problem here?

M4iKK, 2017-09-29:
I want to mention that there is no ISP_FORCE fuse for at32uc3c2512. The code for getfuse and setfuse is not compatible with at32uc3c microcontrollers. If you issue the command `dfu-programmer at32uc3c2512 getfuse ISP_FORCE` you are reading one of two bod33en fuse bits.
I am currently working on a patch for that.

> It seems like for my case it doesn't get set back to 0.

As soon as you issue the launch command ISP_FORCE will be set to 0 by the bootloader. To make sure this is the case you need to read it using a JTAG or aWire programmer as it will be always set to 1 if you try reading it using DFU.

thekenu, 2017-09-29:
@M4iKK This is bootloader version 1.1.0 and there are no fuse bits, instead we have WORD1 and WORD2. This document also explicitly explains that the UC3C has ISP_FORCE in WORD1: http://www.atmel.com/images/doc32166.pdf. 

I also did use JTAG to check my WORD1. And no the ISP_FORCE doesn't get set to 0 by the bootloader, which is the main problem of this issue as I mentioned in the first post.

thekenu, 2017-09-29:
@M4iKK I realized what I was doing wrong the whole time: I never used **launch** after using **flash** . Prior to dfu-programmer, I had been using BatchISP. The flash command for BatchISP would automatically issue a launch command in the end, and I assumed it'd be the same for dfu-programmer. Thanks for point out the launch command. I would have been stuck for another while :-)

closed 2022-10-10 by skirridsystems