dfu-programmer/dfu-programmer #50: Error when opening on Windows 8.1

szszszsz, 2018-01-25:


szszszsz, 2018-01-25:
Sorry, I have created empty issue by mistake (pushing enter does it on title form input). Now the description is updated.

cc @jans23

Zeigren, 2018-02-13:
You shouldn't need the libusb dlls in the same directory as dfu-programmer, you should only need to have the libusb driver installed.

You should be able to update the driver by plugging in your device, hitting reset or whatever on it to have it jump to the bootloader, then go to device manager and it should show up as the name of the micro-controller, update the driver and browse to the dfu-prog-usb-1.2.2 folder.

Hopefully that works and it isn't some other weird problem.

szszszsz, 2018-02-13:
Thank you for the tip! I will let you know once I will check it.

KonstantineF, 2021-01-04:
Just wanted to update this thread to say that @Zeigren you are correct. I had the exact same issues with @szszszsz but I managed to use dfu-programmer by following your solution. The steps were there in the process I followed but I naively tried to use the programmer on its own as some documents mentioned "installing" so I tried to run the .exe file without having it connected to a hardware device (in my case ATMEGA). 

Plugging the device and had to go to Device Manager, and update the drivers but everything run smoothly.

closed 2022-10-10 by skirridsystems