dfu-programmer/dfu-programmer #51: is this abandonware?

GianniGi, 2018-07-05:
I find this program very helpful, and I wonder why it's not updated anymore.  I hope that someone may pick it up! Thanks to the developers for the great work done until now.

swinman, 2018-07-17:
I'm not the maintainer but I've made some updates over time.  I don't have
much extra time to spend on updates that dont directly impact my ability to
use the tool.

szszszsz, 2018-07-17:
@GianniGi 
I would say because it is complete and there is no need ;-)
Check out [other forks](https://github.com/dfu-programmer/dfu-programmer/network) and people working on them - perhaps they would be eager to develop it further and add features you desire.
@swinman thank you for your work!

GianniGi, 2018-07-21:
@swinman thank you! great tool. Too bad that ubuntu 17.10 was delivered with an old version (two years old). Maybe because it was marked as stable and the new one is the instable one?

swinman, 2018-07-21:
actually, dfu-programmer installed on ubuntu is 0.6.1, which has some
functionality missing and is broken for putting stuff on the user page.
the newer (yet now quite old) dfu-programmer 0.7.2 should be much improved
and has tab completion (we routinely use it on ubuntu).  would be great if
they could update to this version..

anthraxx, 2018-11-07:
There is no 0.7.2 tag available on git, can we at least get a tag pushed? :smiley_cat:

wanglouxiaozi, 2020-04-28:
@swinmanIs It convenient for you to add support for STM32F10X series MCU? STM32F10X use the dfu program project(en. Stsw-stm32121 \ stm32_usb-fs-device_lib_v4.1.0 \Projects\Device_Firmware_Upgrade), looking forward to your reply.

wanglouxiaozi, 2020-04-28:
@swinman  Is it convenient for you to add support for STM32F10X series MCU? STM32F10X use the dfu program project(en. Stsw-stm32121 \ stm32_usb-fs-device_lib_v4.1.0 \Projects\Device_Firmware_Upgrade), looking forward to your reply.

swinman, 2020-04-28:
Hi - I don't see any reason why this couldn't be done, but I personally
don't have plans to implement support for this chip.  If you add it, we'll
review and would consider merging but we won't be able to test that it
works without a starter kit.

-Walker

closed 2022-10-10 by skirridsystems

anthraxx, 2022-10-10:
@skirridsystems gentle reminder for tagging 0.7.2

cinderblock, 2022-10-10:
@anthraxx That has been fixed