dfu-programmer/dfu-programmer #52: dfu error how to fix

asciiterminal, 2018-07-31:
root@kali:~/dfu-programmer# make
Making all in src
make[1]: Entering directory '/root/dfu-programmer/src'
make  all-am
make[2]: Entering directory '/root/dfu-programmer/src'
gcc -DHAVE_CONFIG_H -I.    -Wall -g -O2 -MT main.o -MD -MP -MF .deps/main.Tpo -c -o main.o main.c
main.c:29:10: fatal error: usb.h: No such file or directory
 #include <usb.h>
          ^~~~~~~
compilation terminated.
make[2]: *** [Makefile:377: main.o] Error 1
make[2]: Leaving directory '/root/dfu-programmer/src'
make[1]: *** [Makefile:266: all] Error 2
make[1]: Leaving directory '/root/dfu-programmer/src'
make: *** [Makefile:348: all-recursive] Error 1
root@kali:~/dfu-programmer# make install
Making install in src
make[1]: Entering directory '/root/dfu-programmer/src'
gcc -DHAVE_CONFIG_H -I.    -Wall -g -O2 -MT main.o -MD -MP -MF .deps/main.Tpo -c -o main.o main.c
main.c:29:10: fatal error: usb.h: No such file or directory
 #include <usb.h>
          ^~~~~~~
compilation terminated.
make[1]: *** [Makefile:377: main.o] Error 1
make[1]: Leaving directory '/root/dfu-programmer/src'
make: *** [Makefile:348: install-recursive] Error 1

skirridsystems, 2022-10-10:
Closing, as no further input after 4 years

closed 2022-10-10 by skirridsystems

cinderblock, 2022-10-10:
@asciiterminal Please remember to use code blocks for posting code ;)

This is because neither libusb0 nor libusb1 are installed (or are not available) on your system.

`./configure` should probably error if neither of the core dependencies is found.