dfu-programmer/dfu-programmer #53: Add support for at89c51cc03

grimlocker, 2019-01-16:
I was wondering if anyone could add support for the AT89C51CC03 chip? That would be fantastic!

skirridsystems, 2019-01-16:
That chip doesn't appear to have a USB interface, which is what DFU
Programmer works with. You'd need to get the appropriate Atmel tool to
program this device, or maybe there is already a software to support
bootloading over CAN or UART.
Simon

grimlocker, 2019-01-17:
We are currently programming it from Atmel FLIP via USB and an IXXAT dongle, and I wanna get around that somehow due to the old nature of FLIP. So you are saying it can't be done? :(

skirridsystems, 2019-01-17:
Sorry, no it can't be done. DFU programmer uses the Device Firmware Update
protocol over USB. If you look at the FLIP release notes it says that it
uses the IXXAT CAN VCI V3 driver, in other words it uses the specific API
for that one USB-CAN interface.
Simon

grimlocker, 2019-01-17:
That's a shame then.
I do very much appreciate you taking the time to reply though. Thank you!
I suppose I have to keep digging around for a solution.

Best regards, Andy

skirridsystems, 2022-10-05:
Cannot support non-USB chips

closed 2022-10-05 by skirridsystems