dfu-programmer/dfu-programmer #57: exit code is 5 after a successful erase

dwery, 2020-05-24:
dfu-programmer return an exit code of 5 after successfully erasing a chip.

```
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: erase
      quiet: false
      debug: 200
device_type: AVR
------ command specific below ------
   validate: true

...
dfu.c:238: ==============================
dfu.c:240: status->bStatus: errCHECK_ERASED (0x05)
dfu.c:241: status->bwPollTimeout: 0x0000 ms
dfu.c:243: status->bState: dfuERROR (0x0a)
dfu.c:244: status->iString: 0x00
dfu.c:245: ------------------------------
atmel.c:787: Region is NOT bank.
....
atmel.c:795:  First non-blank address in region is 0x0.
atmel.c:866: Flash NOT blank beginning at 0x0.
Not blank at 0x1.
commands.c:88: erase 0x6FFF bytes.
atmel.c:329: atmel_erase_flash( 0x7fff83f43ba0, 4 )
Erasing flash...
....
dfu.c:238: ==============================
dfu.c:240: status->bStatus: errCHECK_ERASED (0x05)
dfu.c:241: status->bwPollTimeout: 0x0000 ms
dfu.c:243: status->bState: dfuERROR (0x0a)
dfu.c:244: status->iString: 0x00
dfu.c:245: ------------------------------
Success
```

`echo $?` -> 5

swinman, 2020-05-24:
Is the chip actually erased --- in other words, is the printing of
'Success' wrong?  5 is DFU_STATUS_ERROR_CHECK_ERASED - which suggests the
chip isn't properly erasing

skirridsystems, 2022-10-10:
Closing, as no further input after 2 years

closed 2022-10-10 by skirridsystems