dfu-programmer/dfu-programmer #59: Trying to use dfu-programmer with at90usb1286

lanceuye, 2021-01-15:
While running in ubuntu I seem to be unable to flash. I am trying to program with `./dfu-programmer at90usb1286 flash TwinkleFox.ino.hex`. 

I am able to program with Flip. And after programming with flip I am able to erase with dfu-programmer. Yay!

However, re-flashing with dfu-programmer gives me an error of the form:

`sudo ./dfu-programmer at90usb1286 flash TwinkleFox.ino.hex 
Checking memory from 0x0 to 0x20FF...  Empty.
0%                            100%  Programming 0x2100 bytes...
[ X  ERROR
Memory write error, use debug for more info.`

I can provide any debug info you ask for, but don't want to flood you with stuff you don't care about. Any help you can provide is appreciated.

swinman, 2021-01-15:
What does the debug log say

lanceuye, 2021-01-15:
sudo ./dfu-programmer at90usb1286 flash --debug 200 TwinkleFox.ino.hex
     target: at90usb1286
    chip_id: 0x2ffb
  vendor_id: 0x03eb
    command: flash
      quiet: false
      debug: 200
device_type: AVR
------ command specific below ------
   validate: true
   hex file: TwinkleFox.ino.hex

dfu.c:332: dfu_device_init(000003eb, 00002ffb)
libusbx: debug [libusb_get_device_list] 
libusbx: debug [libusb_get_device_descriptor] 
dfu.c:347:  0: 0x03eb, 0x2ffb
dfu.c:356: found device at USB:1,9
libusbx: debug [libusb_get_config_descriptor] index 0
dfu.c:605: config 0: maxpower=50*2 mA
dfu.c:612: interface 0
dfu.c:621: setting 0: class:0, subclass 0, protocol:0
dfu.c:634: Found DFU Interface: 0
libusbx: debug [libusb_open] open 1.9
libusbx: debug [usbi_add_pollfd] add fd 11 events 4
dfu.c:367: opened interface 0...
libusbx: debug [libusb_set_configuration] configuration 1
dfu.c:369: set configuration 1...
libusbx: debug [libusb_claim_interface] interface 0
dfu.c:372: claimed interface 0...
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 20000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=0
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0xfad6a8 has callback 0x7fdae4e8c7f0
libusbx: debug [sync_transfer_cb] actual_length=0
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 20000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=0 transferred=6
libusbx: debug [handle_control_completion] handling completion status 0
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0xfad6a8 has callback 0x7fdae4e8c7f0
libusbx: debug [sync_transfer_cb] actual_length=6
dfu.c:238: ==============================
dfu.c:240: status->bStatus: OK (0x00)
dfu.c:241: status->bwPollTimeout: 0x0001 ms
dfu.c:243: status->bState: dfuIDLE (0x02)
dfu.c:244: status->iString: 0x00
dfu.c:245: ------------------------------
dfu.c:696: State: dfuIDLE (2)
atmel.c:1153: atmel_flash( 0x7ffd51310450, 0x7ffd51310330, false, false )
intel_hex.c:673: intel_flash_prep_buffer( 0x7ffd51310330 )
atmel.c:1193: Flash available from 0x0 to 0x1DFFF (64kB p. 0 to 1), 0x1E000 bytes.
atmel.c:1197: Data start @ 0x0: 64kB p 0; 128B p 0x0 + 0x0 offset.
atmel.c:1201: Data end @ 0x20FF: 64kB p 0; 128B p 0x41 + 0x7F offset.
atmel.c:1206: Totals: 0x2100 bytes, 66 128B pages, 1 64kB byte pages.
atmel.c:819: atmel_blank_check( 0x7ffd51310450, 0x00000000, 0x000020FF )
atmel.c:927: atmel_select_memory_unit( 0x7ffd51310450, 0 )
atmel.c:941: Ignore Select Memory Unit for non GRP_AVR32 device.
Checking memory from 0x0 to 0x20FF...  
atmel.c:986: atmel_select_page( 0x7ffd51310450, 0 )
atmel.c:1000: Selecting page 0, address 0x0.
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 20000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
[... 78 lines cut ...]
atmel.c:1424: __atmel_flash_block( 0x7ffd51310450, 0x7ffd51310330, false )
atmel.c:1386: atmel_flash_populate_header( 0x7ffd5130fde0, 0x0, 0x3FF, false )
atmel.c:1340: atmel_flash_populate_footer( 0x7ffd5130fde0, 0x7ffd51310200, 65535, 65535, 65535 )
libusbx: debug [add_to_flying_list] arm timerfd for timeout in 20000ms (first in line)
libusbx: debug [libusb_handle_events_timeout_completed] doing our own event handling
libusbx: debug [handle_events] poll() 4 fds with timeout in 60000ms
libusbx: debug [handle_events] poll() returned 1
libusbx: debug [reap_for_handle] urb type=2 status=-32 transferred=32
libusbx: debug [handle_control_completion] handling completion status -32
libusbx: debug [handle_control_completion] unsupported control request
libusbx: debug [disarm_timerfd] 
libusbx: debug [usbi_handle_transfer_completion] transfer 0xfaecc8 has callback 0x7fdae4e8c7f0
libusbx: debug [sync_transfer_cb] actual_length=32
dfu.c:892: dfu_download ERR: LIBUSB_ERROR_PIPE: Pipe error. 0xfffffff7 (-9)
atmel.c:1485: atmel_flash: flash data dfu_download failed.
atmel.c:1487: Expected message length of 1072, got -9.
atmel.c:1292: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:514: Error writing memory data. (err -4)
libusbx: debug [libusb_release_interface] interface 0
libusbx: debug [libusb_close] 
libusbx: debug [usbi_remove_pollfd] remove fd 11
libusbx: debug [libusb_exit] 
libusbx: debug [libusb_exit] destroying default context

swinman, 2021-01-15:
Sorry dont know whats going on..

lanceuye, 2021-01-15:
Thanks for the quick reply. I'm unclear how to go about further debugging this. Any tips are appreciated.

lanceuye, 2021-02-02:
It turns out I was mistaking the is blank check. Apparently only the first few bytes were erased, so the erase function was succeeding. Use --force during the erase phase got me working.

closed 2021-02-02 by lanceuye