dfu-programmer/dfu-programmer #60: warning: 'libusb_set_debug' is deprecated

ryandesign, 2021-08-30:
```
main.c:81:9: warning: 'libusb_set_debug' is deprecated [-Wdeprecated-declarations]
        libusb_set_debug(usbcontext, debug );
        ^
/opt/local/include/libusb-1.0/libusb.h:1351:1: note: 'libusb_set_debug' has been explicitly marked deprecated here
LIBUSB_DEPRECATED_FOR(libusb_set_option)
^
/opt/local/include/libusb-1.0/libusb.h:70:50: note: expanded from macro 'LIBUSB_DEPRECATED_FOR'
#define LIBUSB_DEPRECATED_FOR(f) __attribute__ ((deprecated))
                                                 ^
1 warning generated.
```

szszszsz, 2021-08-30:
Just a FYI note, this seems to be fixed in one of the forks already:
- https://github.com/zygmunt/dfu-programmer/commit/566da6e955ed013ede4abecb979e10453f4bf987

cinderblock, 2022-10-10:
Fixed in #64

closed 2022-10-10 by cinderblock