dfu-programmer/dfu-programmer #62: Specifying usb-bus and usb-addr does not seem to work on Windows

cinderblock, 2022-09-28:
I'm trying to reliably program two different devices that are connected to my system with two different programs. These two devices are based on the same ATmega32u4 based hardware, so I need to get slightly clever with how I detect the different devices.

I've written a little Node.js script to scan (via [node-usb](https://github.com/node-usb/node-usb)) and filter for the devices I want based on a USB descriptor string (ID `3`).

```js
import { getDeviceList } from "usb";

getDeviceList()
  .filter(d => d.deviceDescriptor.idProduct === 12276)
  .filter(d => d.deviceDescriptor.idVendor === 1003)
  .forEach(device => {
    device.open();

    device.getStringDescriptor(3, (err, desc) => {
      device.close();

      console.log(`Found ${desc} at ${device.busNumber},${device.deviceAddress}`);
    });
  });
```

This gets a "busNumber" and a "deviceAddress" as reported from libusb, as far as I can tell.

Trying to call `dfu-programmer atmega32u4:1,41 erase`, I get "`dfu-programmer: no device present.`", no matter what arguments I've tried. I've even written a little script to walk through all possible bus numbers & device addresses and it's found no matching devices...

Looking into the source a little bit, I see ["found device at ..."](https://github.com/dfu-programmer/dfu-programmer/blob/5254ee489f0048c36f1af660f2354a7746b3f8cc/src/dfu.c#L356) is printed if debug is enabled.

Trying that, without specifying the port, I see:

```log
> dfu-programmer atmega32u4 erase --debug 1000
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: erase
      quiet: false
      debug: 1000
device_type: AVR
------ command specific below ------
   validate: true

dfu.c:414: dfu_device_init( 1003, 12276, 0063FE78, true, false )
dfu.c:431: found device at USB:1,0
dfu.c:663: Found DFU Inteface: 0
dfu.c:293: dfu_abort( 0063FE78 )
[...]
```

So it looks like the device is at "`1,0`" but if I specify a bus or device address of `0`, I always get the "Unsupported target ..." message.

So, is it even possible to use the usb device/address on Windows? I've seen the discussions at #33 about which IDs to use, but those are restricted to Linux. Here, I'm using the numbers reported by the internal debug and it's still not working as I expect.

What am I doing wrong?

swinman, 2022-09-28:
You’ll need the updated libusb 1.. there is a branch where this is
implemented but the makefiles May need some work..

cinderblock, 2022-09-28:
Thank you for the quick response!

Are you saying I need to rebuild `dfu-programmer` with a newer version of libusb 1? Why is this feature included if it's broken on Windows? (Or at least without a warning)

Or maybe I need to switch drivers with (something like) Zadig?

There are so many libusb drivers for Windows!

swinman, 2022-09-28:
It's been a very long time so I'm not sure, but I do remember that the old
libusb doesn't correctly associate device / bus info so it's really hard to
be explicit about which device you are interacting with..

GrantMTG, 2022-09-29:
You need something unique to ID them. In most cases (CDC for example) this is achieved by having a unique serial number. This is part of the device descriptor. Otherwise you have to open each VID+PID combination you want and query the device itself and again you need a way of assigning an ID to the device (EEPROM value, switch or something on the device itself). The purpose of the serial number in the CDC case is to assign a COM port number to the device so that no matter where in the USB tree the device is plugged in, the port # stays.  There might be something to learn from the Mass Storage Device class though. I'm less familiar with that. I don't even remember what class dfu programmer is using. I seem to recall that it's not standard DFU, but I could be wrong.

cinderblock, 2022-09-29:
@GrantMTG Yes, that was the first issue I had to address. But that ended up being easy enough. The bigger issue it seems is that `dfu-programmer` simply isn't able to use the device address on the bus.

----

I'm using LUFA's DFU bootloader with some minor modifications. Notably, I've added a StringDescriptor, with ID `3`, which holds a string unique to each device type that I care to program.

> _USB lets you add an (almost) arbitrary number of general purpose UTF-16 strings.
> **Usually** these strings (numbered `1`, `2`, and `3`) correspond to the ManufacturerString, ProductString, and (optionally) SerialString, as specified in the main Device Descriptor._

The JavaScript program I posted will connect to devices with the correct VID/PID and then print a list of (what I thought were) their USB addresses that libusb uses correlated to that differentiating string. So, I can reliably detect which device is which, even though they share VID & PID.

----

My problem seems to lie in *telling* `dfu-programmer` which device to use. No matter what I've tried, `dfu-programmer` (on Windows) doesn't seem to be able to match to a device. As I've [detailed above](https://github.com/dfu-programmer/dfu-programmer/issues/62#issue-1390019725), it looks like it's not even getting a device ID that makes any sense (always prints that it's working with device `0,1`).

My suspicion is that @swinman is on the right track, that the libusb library we're linking to on Windows builds is missing this functionality. Now to get to the bottom of these complicated libusb libraries...

GrantMTG, 2022-09-29:
Ah OK, thanks for tuning me in. Let us know what gold standard libusb exists. I have a project I did on Ubuntu and TinyCore that I would love to port to Windows, but I see that my PC has a ton of libusb's installed. It's madness.

cinderblock, 2022-09-29:
@GrantMTG I'm currently just downloading pre-built binaries that seem to include a statically linked library - although I'm not **sure** how it's linked... I'm not very familiar with Windows drivers, let alone ones that have been recreated multiple times and the literature references them interchangeably.

I'm looking into building `dfu-programmer` for myself to investigate this. Unfortunately no one has settled on a good open, easy to install, toolchain for Windows. It would be awesome to get some CI scripts working here on GH so that there are always binaries ready to use & try.

cinderblock, 2022-10-03:
Well, I got it building reliably. Also got GitHub Actions working so there is no system ambiguity. This might be useful if anyone else wants to see how to build on Windows or, possibly, as a way to handle distribution builds in the future.

---

Looking into the source, I think I see the problem.

### Address Format

In debug, it's printing "`found device at USB:1,0`", meaning it "found" a device with **device address** of `1` and a **bus address** of `0`, according to [the source](https://github.com/dfu-programmer/dfu-programmer/blob/5254ee489f0048c36f1af660f2354a7746b3f8cc/src/dfu.c#L439). This I think was one of my main confusions as this **order is reversed** from what the [cli expects](https://github.com/dfu-programmer/dfu-programmer/blob/5254ee489f0048c36f1af660f2354a7746b3f8cc/src/arguments.c#L433).

### Bus Address `0` also means "unspecified"

According to [the source](https://github.com/dfu-programmer/dfu-programmer/blob/5254ee489f0048c36f1af660f2354a7746b3f8cc/src/dfu.c#L434) and [comments](https://github.com/dfu-programmer/dfu-programmer/blob/5254ee489f0048c36f1af660f2354a7746b3f8cc/src/arguments.h#L162), `0` means the address was not specified on the command line and `dfu_device_init(...)` just picks the first device.

Since many device are on USB bus number `0`, they'll never be specifiable as a targeted programing.

### Fix

I've created a fix for this that [increments the `bus_id` by `1`](https://github.com/cinderblock/dfu-programmer/commit/e0048b11fc897f2d6972c44547a1b8f4b89771bc) when input and used. This keeps the internal definition of a bus ID of `0` meaning use the first found device but lets a user specify a bus ID of "`0`" on the CLI _(which internally is stored as a `1`)_.

### Get All IDs

Now to find a way to lookup what these IDs are in a reliable way... They don't match the IDs produced by my `node-usb` based lookup program, probably because they use different versions (or implementations) of the usb library.

cinderblock, 2022-10-05:
This would be fixed by #64

closed 2022-10-21 by cinderblock