dfu-programmer/dfu-programmer #63: Build on Windows

cinderblock, 2022-10-02:
I'm trying to debug #62 by building my own copy of `dfu-programmer.exe`. I tried plenty of things on my personal machine and kept running into issues with paths, msys, cygwin, tried in multiple bash shells.

I'm now [trying to build](https://github.com/cinderblock/dfu-programmer/actions) on GitHub Actions. It was easy enough to get macOS and Linux building but Windows is still giving no end of trouble.

I've tried the instructions in the `README` and the `release_proc.txt` files to no avail. Any better instructions out there? How are the releases normally handled?

cinderblock, 2022-10-03:
Well, after a lot of struggling with msys/cygwin paths, 32 vs 64 bit versions, library paths, silent by default errors, I've got it building and apparently running (`--version` works and it loads/needs the `libusb0.dll`) from Atmel FLIP. I'll post more details here when I get around to it.

I'm hoping to make my repo have automatic publishing of binaries making any updates easy to push out for anyone.

cinderblock, 2022-10-04:
I've got libusb1 based building targeting x64 machines now too.

https://github.com/cinderblock/dfu-programmer/actions/runs/3178481630

Any appetite here for an update?

GrantMTG, 2022-10-04:
Excellent. Do we have the source and/or a way to build libusb0, libusb1 for Windows? Or what is the recommended repo?

cinderblock, 2022-10-05:
I'm using the MSYS2 packages `mingw-w64-i686-libusb` and `mingw-w64-x86_64-libusb` for [x86](https://github.com/cinderblock/dfu-programmer/blob/libusb1/.github/workflows/build.yml#L148) and [x64](https://github.com/cinderblock/dfu-programmer/blob/libusb1/.github/workflows/build.yml#L239) builds respectively. They publish statically linkable libraries that link to dlls that they also publish.

I've [chosen to include those dlls](https://github.com/cinderblock/dfu-programmer/blob/libusb1/.github/workflows/build.yml#L183) in the "release" zip that I'm [publishing on my fork](https://github.com/cinderblock/dfu-programmer/releases). (I might need to include a copy of `COPYING`...)

I wonder if it would be possible to statically include `libusb1` to remove all external dependencies.

closed 2022-10-10 by cinderblock