dfu-programmer/dfu-programmer #68: dfu.c:961: Unknown error 0xfffffff7 (-9)

cinderblock, 2022-10-21:
issue with the ATXMEGA128a3u. 
Error message:

```log
dfu.c:961: Unknown error 0xfffffff7 (-9)
```

We've tested this on Linux ARM (dfu-programmer versions 0.6.1 - 0.7.2) and Linux x86 environment (Ubuntu, dfu-programmer version 0.6.1).
Application 'Flip' is working fine.

We're calling the application in following way:

```bash
DEBUGLEVEL=999999
TARGET=atxmega128a3u
FW_FILE=DL_3.50_640T_V1.5.hex
#FW_FILE=FW_3v51_ErrorLogger.hex

sudo dfu-programmer $TARGET erase --debug $DEBUGLEVEL
echo "------------------------------------------"
echo "Return from erasing: $?"
echo "------------------------------------------"

sudo dfu-programmer $TARGET flash $FW_FILE --debug $DEBUGLEVEL
echo "------------------------------------------"
echo "Return from flashing: $?"
echo "------------------------------------------"
sudo dfu-programmer $TARGET start
```

Attached the extended log file
[dfu-programmer.ext.debug.log](https://github.com/dfu-programmer/dfu-programmer/files/1803421/dfu-programmer.ext.debug.log)

Can somebody help us?
Best Regards,
Helmut

_Originally posted by @SaubaSogI in https://github.com/dfu-programmer/dfu-programmer/issues/29#issuecomment-372355274_

cinderblock, 2022-10-25:
@SaubaSogI Can you reproduce with a single run of the command that reliably produces the error on the latest version? It has better debug outputs and is guaranteed to be based on the newer libusb1.

SaubaSogI, 2022-10-25:
@cinderblock: We've stopped this project and I cannot reproduce this any more, I'm really sorry.

cinderblock, 2022-10-25:
No worries. It's been a while since anyone has worked on this project.

closed 2022-10-25 by cinderblock