dfu-programmer/dfu-programmer #83: atmega32u4 - error while flashing

stephendpmurphy, 2023-01-03:
Hi :wave:

I can't seem to get any application to flash via DFU. I am using a custom board w/ the atmega32u4-MU MCU and I am using the USB-DFU .hex image provided by Microchip on their website. You can find it at the very bottom of the page in the ```Embedded Software``` section on the [summary page for the ATmega32u4](https://www.microchip.com/en-us/product/ATmega32U4#document-table).

Once I flash that image, I get what appears to be a working DFU bootloader to enumerate
```console
[356247.965362] usb 3-4.2.2.4.4: new full-speed USB device number 125 using xhci_hcd
[356248.072389] usb 3-4.2.2.4.4: New USB device found, idVendor=03eb, idProduct=2ff4, bcdDevice= 0.00
[356248.072402] usb 3-4.2.2.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[356248.072406] usb 3-4.2.2.4.4: Product: ATm32U4DFU
[356248.072409] usb 3-4.2.2.4.4: Manufacturer: ATMEL
[356248.072411] usb 3-4.2.2.4.4: SerialNumber: 1.0.0
```

When I attempt to flash a test .hex image that turns on an LED, I get the following error.
```
⚡ ./dfu-programmer atmega32u4 flash NEW_PROJECT-v0_1_0.hex --debug 5
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: flash
      quiet: false
      debug: 5
device_type: AVR
------ command specific below ------
   validate: true
   hex file: NEW_PROJECT-v0_1_0.hex

Checking memory from 0x0 to 0xFF...  Empty.
0%                            100%  Programming 0x100 bytes...
[ X  ERROR
Memory write error, use debug for more info.
```

My immediate assumption is that this DFU firmware I found is not compatible with ```dfu-programmer```. If that is the case, do you have any .hex images for compatible DFU bootloaders I could try? I saw mention in the docs that Atmel/Microchip provides chips pre-loaded with a boot-loader, but none of my ICs so far have had one.

Thanks!

cinderblock, 2023-01-03:
Hmm, this looks to me like `dfu-programmer` is loading the `.hex` file correctly but having trouble sending the data to the target device.

What version of `dfu-programmer` are you using? `./dfu-programmer --version`.

Do you have the same error when running under `sudo`? Maybe it is a perimission/udev rule issue?

Can you run the same thing with much larger debug setting: `./dfu-programmer atmega32u4 flash my.hex --debug 10000`

Note, according to the datasheet, under "Ordering Information", only the `-AU` (TQFP) versions include the bootloader from the factory.

----

Relevant is that the official DFU bootloader from Atmel automatically detects 8MHz vs 16MHz cpu clock at runtime (by comparing to WDT), so that shouldn't be affecting anything. (Also your device enumerates so that seems to be working fine).

From a disassembly of the official DFU bootloader firmware:

```asm
7240:	10-F0       	brbs 0, .+4           ; Branch if Bit in SREG is Set            # if (Carry set) branch(+4)
7242:	02-E1       	ldi r16, 0x12         ; Load Immediate                          # r16 = 0x12   // With PLLCSR.PINDIV set
7244:	01-C0       	rjmp .+2              ; Relative Jump                           # skip next
7246:	02-E0       	ldi r16, 0x02         ; Load Immediate                          # r16 = 0x02   // With PLLCSR.PINDIV cleared
7248:	09-BD       	out 0x29, r16         ; Store Register to I/O Location          # PLLCSR = r16
724A:	08-95       	ret                   ; Return from Subroutine                  # 
```

stephendpmurphy, 2023-01-04:
Hey thanks for getting back to me so quickly.. Here is the information as requested.

- dfu-programmer version:
  ```console
  ⚡ ./dfu-programmer --version                                        
  dfu-programmer 1.0.0
  ```

- Adding ```sudo``` unfortunately didn't help and I get the same error. (I had high hopes when I read that since I usually take udev for granted.)

- The DFU image I am using is [here](https://github.com/greatlakesdev/megaAVR_USB_DFU_firmware/blob/main/ATMega32U4-usbdevice_dfu-1_0_0.hex)

- Here is the output with the debug level cranked.
```console
⚡ sudo ./dfu-programmer atmega32u4 flash NEW_PROJECT-v0_1_0.hex --debug 10000
     target: atmega32u4
    chip_id: 0x2ff4
  vendor_id: 0x03eb
    command: flash
      quiet: false
      debug: 10000
device_type: AVR
------ command specific below ------
   validate: true
   hex file: NEW_PROJECT-v0_1_0.hex

dfu.c:318: dfu_device_init( 1003, 12276, 0x7ffe71f591d0, true, false )
dfu.c:322: dfu_device_init(000003eb, 00002ff4)
dfu.c:336:  0: 0x1d6b, 0x0003
dfu.c:336:  1: 0x0c45, 0x672a
dfu.c:336:  2: 0x2109, 0x8818
dfu.c:336:  3: 0x046d, 0xc52b
dfu.c:336:  4: 0x0bda, 0x48f0
dfu.c:336:  5: 0x03eb, 0x2ff4
dfu.c:346: found device at USB:3,13
dfu.c:502: dfu_find_interface()
dfu.c:512: config 0: MaxPower=50*2 mA
dfu.c:519: interface 0
dfu.c:526: setting 0: class:0, subclass 0, protocol:0
dfu.c:541: Found DFU Interface: 0
dfu.c:357: opened interface 0...
dfu.c:359: set configuration 1...
dfu.c:362: claimed interface 0...
dfu.c:292: dfu_abort( 0x7ffe71f591d0 )
dfu.c:203: dfu_get_status( 0x7ffe71f591d0, 0x7ffe71f59120 )
dfu.c:229: ==============================
dfu.c:230: status->bStatus: OK (0x00)
dfu.c:232: status->bwPollTimeout: 0x0001 ms
dfu.c:233: status->bState: dfuIDLE (0x02)
dfu.c:235: status->iString: 0x00
dfu.c:236: ------------------------------
dfu.c:569: State: dfuIDLE (2)
atmel.c:1151: atmel_flash( 0x7ffe71f591d0, 0x7ffe71f590d0, false, false )
intel_hex.c:675: intel_flash_prep_buffer( 0x7ffe71f590d0 )
atmel.c:1189: Flash available from 0x0 to 0x6FFF (64kB p. 0 to 0), 0x7000 bytes.
atmel.c:1194: Data start @ 0x0: 64kB p 0; 128B p 0x0 + 0x0 offset.
atmel.c:1198: Data end @ 0xFF: 64kB p 0; 128B p 0x1 + 0x7F offset.
atmel.c:1202: Totals: 0x100 bytes, 2 128B pages, 1 64kB byte pages.
atmel.c:819: atmel_blank_check( 0x7ffe71f591d0, 0x00000000, 0x000000FF )
atmel.c:927: atmel_select_memory_unit( 0x7ffe71f591d0, 0 )
atmel.c:941: Ignore Select Memory Unit for non GRP_AVR32 device.
Checking memory from 0x0 to 0xFF...  
atmel.c:986: atmel_select_page( 0x7ffe71f591d0, 0 )
atmel.c:999: Selecting page 0, address 0x0.
dfu.c:144: dfu_download( 0x7ffe71f591d0, 4, 0x7ffe71f58fc3 )
dfu.c:165: Message: m[0] = 0x06
dfu.c:165: Message: m[1] = 0x03
dfu.c:165: Message: m[2] = 0x00
dfu.c:165: Message: m[3] = 0x00
dfu.c:203: dfu_get_status( 0x7ffe71f591d0, 0x7ffe71f58fb4 )
[... 334 lines cut ...]
dfu.c:165: Message: m[288] = 0x00
dfu.c:165: Message: m[289] = 0x00
dfu.c:165: Message: m[290] = 0x00
dfu.c:165: Message: m[291] = 0x00
dfu.c:165: Message: m[292] = 0x10
dfu.c:165: Message: m[293] = 0x44
dfu.c:165: Message: m[294] = 0x46
dfu.c:165: Message: m[295] = 0x55
dfu.c:165: Message: m[296] = 0x01
dfu.c:165: Message: m[297] = 0x10
dfu.c:165: Message: m[298] = 0xff
dfu.c:165: Message: m[299] = 0xff
dfu.c:165: Message: m[300] = 0xff
dfu.c:165: Message: m[301] = 0xff
dfu.c:165: Message: m[302] = 0xff
dfu.c:165: Message: m[303] = 0xff
dfu.c:694: dfu_download ERR: LIBUSB_ERROR_PIPE: Pipe error. 0xfffffff7 (-9)
atmel.c:1485: atmel_flash: flash data dfu_download failed.
atmel.c:1486: Expected message length of 304, got -9.
atmel.c:1292: Error flashing the block: err -2.
ERROR
Memory write error, use debug for more info.
commands.c:529: Error writing memory data. (err -4)

```

cinderblock, 2023-01-04:
Thanks for this. I'll look into the code.

Did this happen on recent versions? We recently switched to forcing libusb1. It seems there were situations before v1.0.0 where, if both libusb1 and libusb0 where available, it would default to the latter. So this may be a latent bug that has just now surfaced.

---

Separately, I'm noodling on some dedicated hardware to run unit tests. I'm wondering if it would be ok for me to add this hex as a test to make sure it programs.

cinderblock, 2023-01-04:
Is there a chance #59 has your solution? Run `./dfu-programmer atmeag32u4 erase --force` first?

stephendpmurphy, 2023-01-05:
I have no issue with you keeping that Hex or the others in that repo as a test subject. I just pulled them from Microchips site and stored them in a git repo in case Microchip pulled them down. I'll look into #59 here shortly and report back.

stephendpmurphy, 2023-01-06:
Ok. So doing the ```./dfu-programmer atmeag32u4 erase --force``` did work. I was able to ```flash``` after that. I guess I'm unclear as to what the problem is? Immediately after flashing, I attempted to flash again and got a bit clearer error message

```console
⚡ ./dfu-programmer atmega32u4 flash NEW_PROJECT-v0_1_0.hex
Checking memory from 0x0 to 0xFF...  Not blank at 0x1.
The target memory for the program is not blank.
Use --force flag to override this error check.
```

Ok sure.. Flash isn't empty, so if I try to use the ```--force``` flag as the error indicates I should.. I get another generic error with no context.

```console
⚡ ./dfu-programmer atmega32u4 flash NEW_PROJECT-v0_1_0.hex --force  
0%                            100%  Programming 0x100 bytes...
[ X  ERROR
Memory write error, use debug for more **info.**
```

Is this all expected behavior? From my perspective there's almost ALWAYS going to be something at the start of flash, so is my process to just always force the erase and then flash my image? Or is there a problem that needs to be address in the tool?

All that being said, thank you for your help getting me to this point! I'm super excited. Just FYI, my intention to is to write a sort of "bootstrap" bootloader that sits just before the DFU bootloader and checks for a valid application via a checksum. If no valid application is found at the start of flash then we jump to the DFU bootloader at the end of flash.

cinderblock, 2023-01-06:
I'm a new maintainer here and this is the kind of thing I'd like to clean up. Now that we have a clear test case, we can work on a cleaner fix. Thank you for reporting this annoyance.

---

There are a lot of strategies for this. You'll need a custom bootloader to change the entry point to you extra section. At that point, you might as well build the functionality into your bootloader. The only thing that is special about that space are the lock-bits.

cinderblock, 2023-01-22:
Might be related so I'll leave this here for now.

<details>
<summary>Initial <code>read</code> doesn't work</summary>

```log
action@AVRActionPi:~/runner/_work/dfu-programmer/dfu-programmer $ src/dfu-programmer at90usb82 read --debug 10000
     target: at90usb82
    chip_id: 0x2ff7
  vendor_id: 0x03eb
    command: read
      quiet: false
      debug: 10000
device_type: AVR
------ command specific below ------

dfu.c:317: dfu_device_init( 1003, 12279, 0x7ff24da910, true, false )
dfu.c:321: dfu_device_init(000003eb, 00002ff7)
dfu.c:335:  0: 0x1d6b, 0x0003
dfu.c:335:  1: 0x03eb, 0x2ff7
dfu.c:346: found device at USB:1,49
dfu.c:522: dfu_find_interface()
dfu.c:532: config 0: MaxPower=50*2 mA
dfu.c:539: interface 0
dfu.c:546: setting 0: class:0, subclass 0, protocol:0
dfu.c:561: Found DFU Interface: 0
dfu.c:360: opening interface 0...
dfu.c:364: returned 0...
dfu.c:371: opened interface 0...
dfu.c:380: set configuration 1...
dfu.c:389: claimed interface 0...
dfu.c:290: dfu_abort( 0x7ff24da910 )
dfu.c:201: dfu_get_status( 0x7ff24da910, 0x7ff24da810 )
dfu.c:227: ==============================
dfu.c:228: status->bStatus: OK (0x00)
dfu.c:230: status->bwPollTimeout: 0x0001 ms
dfu.c:231: status->bState: dfuIDLE (0x02)
dfu.c:233: status->iString: 0x00
dfu.c:234: ------------------------------
dfu.c:589: State: dfuIDLE (2)
atmel.c:640: atmel_read_flash( 0x7ff24da910, 0x7ff24da868, 0, false )
atmel.c:927: atmel_select_memory_unit( 0x7ff24da910, 0 )
atmel.c:941: Ignore Select Memory Unit for non GRP_AVR32 device.
Reading 0x1000 bytes...
atmel.c:986: atmel_select_page( 0x7ff24da910, 0 )
atmel.c:999: Selecting page 0, address 0x0.
dfu.c:142: dfu_download( 0x7ff24da910, 4, 0x7ff24da7d0 )
dfu.c:163: Message: m[0] = 0x06
dfu.c:163: Message: m[1] = 0x03
dfu.c:163: Message: m[2] = 0x00
dfu.c:163: Message: m[3] = 0x00
dfu.c:201: dfu_get_status( 0x7ff24da910, 0x7ff24da7e0 )
dfu.c:227: ==============================
dfu.c:228: status->bStatus: OK (0x00)
dfu.c:230: status->bwPollTimeout: 0x0001 ms
dfu.c:231: status->bState: dfuIDLE (0x02)
dfu.c:233: status->iString: 0x00
dfu.c:234: ------------------------------
dfu.c:142: dfu_download( 0x7ff24da910, 6, 0x7ff24da7e0 )
dfu.c:163: Message: m[0] = 0x03
dfu.c:163: Message: m[1] = 0x00
dfu.c:163: Message: m[2] = 0x00
dfu.c:163: Message: m[3] = 0x00
dfu.c:163: Message: m[4] = 0x03
dfu.c:163: Message: m[5] = 0xff
dfu.c:714: dfu_download ERR: LIBUSB_ERROR_PIPE: Pipe error. 0xfffffff7 (-9)
atmel.c:602: dfu_download failed
atmel.c:712: Error reading block 0x0 to 0x3FF: err -1.
ERROR
Memory read error, use debug for more info.
[... 318 lines cut ...]
dfu.c:163: Message: m[5] = 0xff
dfu.c:177: dfu_upload( 0x7fd389d340, 1024, 0x5596f36200 )
dfu.c:142: dfu_download( 0x7fd389d340, 6, 0x7fd389d210 )
dfu.c:163: Message: m[0] = 0x03
dfu.c:163: Message: m[1] = 0x00
dfu.c:163: Message: m[2] = 0x08
dfu.c:163: Message: m[3] = 0x00
dfu.c:163: Message: m[4] = 0x0b
dfu.c:163: Message: m[5] = 0xff
dfu.c:177: dfu_upload( 0x7fd389d340, 1024, 0x5596f36600 )
dfu.c:142: dfu_download( 0x7fd389d340, 6, 0x7fd389d210 )
dfu.c:163: Message: m[0] = 0x03
dfu.c:163: Message: m[1] = 0x00
dfu.c:163: Message: m[2] = 0x0c
dfu.c:163: Message: m[3] = 0x00
dfu.c:163: Message: m[4] = 0x0f
dfu.c:163: Message: m[5] = 0xff
dfu.c:177: dfu_upload( 0x7fd389d340, 1024, 0x5596f36a00 )
Success
Memory is blank, returning a single blank page.
Use --force to return the entire memory regardless.
Dumping 0x80 bytes from address offset 0x0.
:00000001FF
```
</details>

cinderblock, 2023-01-25:
I've got a new test branch setup just to track this issue: https://github.com/dfu-programmer/dfu-programmer/tree/test-read-after-reset

You can see a reproducible test here: https://github.com/dfu-programmer/dfu-programmer/actions/runs/3995365828/jobs/6854154470

Shown are the log outputs from running `erase --force` before `read` and not. It **should not be necessary** to erase before reading...

This is currently only being tested on an empty AVR. I wonder if this is different if the AVR is not empty...