dfu-programmer/dfu-programmer #86: AT90USB64 (Retrode 2) is not recognised

PascalLeroi, 2023-01-10:
Might be that I'm doing something wrong, but I can't seem to connect my Retrode 2 (Which it seems to have an AT90USB64 in it)
It just keeps saying that the device is not present.
I've tried with a 6 and a 7 at the end, as otherwise the program will just give me a list of compatible devices.
[image]
Am I missing something or is this device not supported?
[image]
[image]
[image]
Thanks for taking the time to look at this :)

cinderblock, 2023-01-10:
The "⚠️" icon in device manager indicates that the USB device is not working as Windows expects and probably won't connect.

I **think** this is a driver issue.

Try running Zadig and see what driver it is using for the device?

If you only have one target installed, you shouldn't need the `:4,2` at the end of the target.

Dallagen2, 2023-02-09:
Hi, this is an issue with Windows 8 and onwards as well as a mistake on Microchip's part. 

Here's a full detailed breakdown of the issue, with some solutions. https://fjlaboratories.com/blog/usb-enumeration-windows

For a short rundown:

Microchip mistakenly reports the USB version of the AT90USB64 bootloader as USB 20.00, instead of USB 02.00, which makes windows ask it for further identifying information that was added to usb spec in 2.1 onwards, as this behavior is not expected or supported by the DFU bootloader, it does not respond, and in turn windows refuses to enumerate the device, as it believes it is malfunctioning.

This is solved by using the registry keys added by microsoft 2 weeks after windows 8.1 release that allow you to bypass the BOS request for specific VID/PID/REV combinations on usb devices:

For AT90USB64_ devices, copy and paste the following lines into Windows Command Prompt as an administrator:

```
reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\03EB2FF90000" /v "osvc" /t REG_BINARY /d "0000" /f
reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\03EB2FF90000" /v "SkipContainerIdQuery" /t REG_BINARY /d "01000000" /f
reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\03EB2FF90000" /v "SkipBOSDescriptorQuery" /t REG_BINARY /d "01000000" /f
```

For AT90USB128_ devices, copy and paste the following lines into Windows Command Prompt as an administrator:

```
reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\03EB2FFB0000" /v "osvc" /t REG_BINARY /d "0000" /f
reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\03EB2FFB0000" /v "SkipContainerIdQuery" /t REG_BINARY /d "01000000" /f
reg add "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\03EB2FFB0000" /v "SkipBOSDescriptorQuery" /t REG_BINARY /d "01000000" /f
```

This can also be solved by flashing a new bootloader onto it with an ISP programmer, as recommended by Microchip in their support article referencing the issue here: https://techcommunity.microsoft.com/t5/microsoft-usb-blog/why-does-my-usb-device-work-on-windows-8-0-but-fail-on-windows-8/ba-p/270819