dfu-programmer/dfu-programmer #93: Release asset dfu-programmer-1.1.0.tar.gz is missing most header files

the2masters, 2023-07-08:
Hi,
I tried to compile dfu-programmer-1.1.0.tar.gz from releases page, but many header files are missing:
src/libdfu.h
src/arguments.h
src/dfu-device.h
src/atmel.h
src/intel_hex.h
src/dfu.h
src/commands.h
src/util.h
src/stm32.h

Can you check this? Thanks!

cinderblock, 2023-07-08:
Thank you for the report. Looking into this. I thought we had all the automatic publishing tools working...

cinderblock, 2023-07-08:
I believe this is now fixed. Sorry!

the2masters, 2023-07-08:
It's getting better, now only src/libdfu.h is missing :-)

cinderblock, 2023-07-08:
Derp! I knew I was forgetting something. Time to add a test of the built archive before publish lol

cinderblock, 2023-07-08:
f23d6dfd6671b46cbf1279a61291f6fa18d7559e and dcb628c4166ffc012e5bcf8c25c2e32708a3fad4 through e064ea50d1239f8529192c71bcb95a218a1e2d3a should do it and keep it stable :)

closed 2023-07-08 by cinderblock

the2masters, 2023-07-08:
Now it's fixed, thank you! I'm going to update the openwrt package so people can install it on their router and update their DIY devices connected to the router's USB port.... Probably I am the only user of this crazy setup :-D

cinderblock, 2023-07-09:
Sounds fun! If that's the only USB port you've got or you have some custom device, great!

commit 069fcf36 (2023-07-08): Re-add mistakenly removed header files from source list. Fixes #93