dfu-programmer/dfu-programmer #94: dfu-programmer 1.1.0 build failure

chenrui333, 2023-07-08:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>error build log</summary>

```
  main.c:22:10: fatal error: 'libdfu.h' file not found
  #include "libdfu.h"
           ^~~~~~~~~~
  commands.c:28:10: fatal error: 'commands.h' file not found
  #include "commands.h"
           ^~~~~~~~~~~~
  intel_hex.c:39:10: fatal error: 'intel_hex.h' file not found
  #include "intel_hex.h"
           ^~~~~~~~~~~~~
  1 error generated.
  make[1]: *** [main.o] Error 1
  make[1]: *** Waiting for unfinished jobs....
  stm32.c:27:10: fatal error: 'dfu-device.h' file not found
  #include "dfu-device.h"
           ^~~~~~~~~~~~~~
  atmel.c:31:10: fatal error: 'dfu-device.h' file not found
  #include "dfu-device.h"
           ^~~~~~~~~~~~~~
  arguments.c:28:10: fatal error: 'dfu-device.h' file not found
  #include "dfu-device.h"
           ^~~~~~~~~~~~~~
  dfu.c:30:10: fatal error: 'dfu.h' file not found
  #include "dfu.h"
           ^~~~~~~
  libdfu.c:25:10: fatal error: 'dfu-device.h' file not found
  #include "dfu-device.h"
           ^~~~~~~~~~~~~~
  1 error generated.
  1 error generated.
  1 error generated.
  1 error generated.
```

</details>

full build log, https://github.com/Homebrew/homebrew-core/actions/runs/5492765143/jobs/10012914104?pr=136121
relates to Homebrew/homebrew-core#136121

cinderblock, 2023-07-08:
Looks like this is a duplicate of #93. Please track that issue instead. Please re-open this issue if you think it is a different issue.

closed 2023-07-08 by cinderblock

cinderblock, 2023-07-08:
I think this is fixed now.