dfu-programmer/dfu-programmer #95: Update handling of `docs/*` and `fedora/*`

cinderblock, 2023-07-08:
With the new GitHub release procedure, I would like to minimize the number of manual things developers need to remember to do.

High on that list is keeping a few files up to date automatically. This is a place to discuss any such changes and impacts they might have on software ecosystems.

 - `docs/dfu-programmer.1`
 - `docs/dfu-programmer.html`
 - `fedora/dfu-programmer.spec.in`

Erovia, 2024-01-01:
I'm not sure if any of the files under the `fedora` directory provide any value.

The SPEC file is outdated and is different from the one Fedora used to package.

I'm not entirely sure what the purpose of the FDI file was, but it was not used by the last version of the SPEC file used by Fedora maintainers. 
In the standard repos of Fedora 38 and 39 (the two currently maintained versions) only 1 package uses the  `/usr/share/hal/` directory.
From what I gather, `udev` replaced `hal`.

Fedora stopped packaging `dfu-programmer` with release 39 (released on 2023-11-06).
Although an official package is still available for 38 (EOL 2024-05-14), it's an old version: `0.7.2`.

As I occasionally use `dfu-programmer`, I created a Copr project to package and host up-to-date packages for current Fedora releases:
https://copr.fedorainfracloud.org/coprs/erovia/dfu-programmer/

cinderblock, 2024-01-08:
> I'm not sure if any of the files under the `fedora` directory provide any value.

I agree but I'm not a Fedora person so I have little reference to know what the right set of extra files is. 

> The SPEC file is outdated and is different from the one Fedora used to package.

Yeah, I didn't want to commit to maintaining that without more details. Kinda just ignored them.

> I'm not entirely sure what the purpose of the FDI file was, but it was not used by the last version of the SPEC file used by Fedora maintainers. In the standard repos of Fedora 38 and 39 (the two [...]

Some more "offficial" guides on what is needed might be nice.

> Fedora stopped packaging `dfu-programmer` with release 39 (released on 2023-11-06). Although an official package is still available for 38 (EOL 2024-05-14), it's an old version: `0.7.2`.

So Fedora declined to include the latest changes? Any indication about why?

> As I occasionally use `dfu-programmer`, I created a Copr project to package and host up-to-date packages for current Fedora releases: https://copr.fedorainfracloud.org/coprs/erovia/dfu-programmer/

👍

Erovia, 2024-01-08:
> > I'm not sure if any of the files under the `fedora` directory provide any value.
> 
> I agree but I'm not a Fedora person so I have little reference to know what the right set of extra files is.
> 
Imho, no extra files should be necessary.
Optionally, you can include a SPEC file in the root of the repo to help potential packagers with a starting point.
Sure you can create OS-specific directories or a perhaps a `SPECS` directory under which one can contribute distro-specific files (e.g.: `SPECS/fedora.spec`, `SPECS/opensuse.spec`, etc).
On one hand, this could help with adoption of new releases, on the other it puts maintenance burden on you and other contributors of this repository.

> > The SPEC file is outdated and is different from the one Fedora used to package.
> 
> Yeah, I didn't want to commit to maintaining that without more details. Kinda just ignored them.
> 
None of the files currently under the `fedora/` dir was used by the Fedora packagers in the last release that carried `dfu-programmer`: <https://src.fedoraproject.org/rpms/dfu-programmer/tree/f38>

> Some more "offficial" guides on what is needed might be nice.
I digged into it a bit more and HAL was deprecated in Fedora 16 (2011): <https://fedoraproject.org/wiki/Features/HalRemoval>
The Fedora packagers stopped including the HAL policy in 2017: <https://src.fedoraproject.org/rpms/dfu-programmer/c/c441e026a31cb03c4bb8cda766cf3b019716e04e?branch=rawhide>

> > Fedora stopped packaging `dfu-programmer` with release 39 (released on 2023-11-06). Although an official package is still available for 38 (EOL 2024-05-14), it's an old version: `0.7.2`.
> 
> So Fedora declined to include the latest changes? Any indication about why?
> 
The latest release packaged by Fedora for F38 was `0.7.2`. <https://src.fedoraproject.org/rpms/dfu-programmer>
At some point, it seems the package was no longer actively maintained and as per [their packaging policy](https://docs.fedoraproject.org/en-US/fesco/Policy_for_orphan_and_retired_packages/) it was first retired than orphaned.
This is why it was not packaged for the latest release (F39).

---

I'm happy to contribute my SPEC file which was based on the one used by Fedora for F38.
It builds and packages the `1.1.0` release without any patches on F38, F39 and the development branch.
It can probably do the same on Fedora-based distros as well and potentially other RPM-based distros. (some changes might be necessary to make it more distro-agnostic)

cinderblock, 2024-01-10:
That all sounds good to me. Feel free to open a PR.

Keep in mind, I put a high value on minimizing the number of secondary things that a developer needs to do to keep all repo files updated.

Erovia, 2024-01-12:
So what's your preferred approach would be?

cinderblock, 2024-01-15:
I guess it makes the most sense to just get rid of the vestigial files. I don't want to be maintaining many OS specific files, especially for ones I don't use.

Erovia, 2024-01-20:
Created PR https://github.com/dfu-programmer/dfu-programmer/pull/98 to remove the `fedora` dir with its content and to place my SPEC file into the root of the repo.